oculus-samples/Unity-PassthroughCameraApiSamples #3: Demos don't work

zcha621, 2025-03-19:
1. I updated my Quest 3 OS to V74
2. Then I factory reset my headset
3. I cloned the repository, built the APK in Unity 2022.3.60f1
4. Install the app on quest 3 and give all required permission.
5. However, the demo doesn't work. For example, the CameraViewer demo showed a white plane underneath labeled 'permission granted'. I assume I could see the camera view but actually not.

More details of the test can be found in the following recorded video:

[https://youtube.com/shorts/lJLUCGHAXKg?feature=share]

zsp1987, 2025-03-21:
When you build the app, did you see any error message?

zcha621, 2025-03-21:
I didn't see any error.

FYI, this issue relates to issue #2; I had issue #2 and then conducted steps 1 and 2 in this issue.

JordiFB, 2025-03-27:
@zcha621  After solve the installation issue #2 , Are you still having problems getting the camera image?

Could you update this thread with your current status and , if the issue is still present, could you share some logs with us?

Thank you

zcha621, 2025-03-27:
Thank you for your response!

I hadn't touched the app for several days since my last failed attempt to run the demo.

However, I tried running it again today without making any changes, and surprisingly, it worked!

https://www.youtube.com/shorts/mDRfcb6w908

closed 2025-03-27 by zcha621

XuanbingXie, 2025-05-02:
I encountered the same problem as you. How did you solve it in the end? @zcha621