oculus-samples/Unity-PassthroughCameraApiSamples #4: Can't get horizonos.permission.HEADSET_CAMERA permission

NewbeeFan, 2025-03-21:
Thank you for providing excellent example.

I am trying to get the Passthrough camera data. However, I have benn stucked by how to get horizonos.permission.HEADSET_CAMERA permission.

I put the permission request as blow, but the app in quest3 net asked for the horizonos.permission.HEADSET_CAMERA permission.
[image]

JordiFB, 2025-03-21:
Hi @NewbeeFan , thanks for conact us.

Could you give us more details? Which Horizon Os version are you using and Unity Engine version. 

Are you experimenting this problem with our samples code or in your own project?

Remember that access to Passthrough Camera data is available in Horizon OS v74.

NewbeeFan, 2025-03-21:
> Hi [@NewbeeFan](https://github.com/NewbeeFan) , thanks for contact us.
> 
> Could you give us more details? Which Horizon Os version are you using and Unity Engine version.
> 
> Are you experimenting this problem with our samples code or in your own project?
> 
> Remember that access to Passthrough Camera data is available in Horizon OS v74.

Thank you for reply!
OS is V74. Unity Engine is 2022.3.58f1.
I just have solved the problem. I added the following codes in AndroidManifest.xml and it worked.
  <uses-permission android:name="android.permission.CAMERA" />
  <uses-permission android:name="android.permission.ACCESS_MEDIA_LOCATION" />
  <uses-feature android:name="android.hardware.camera" android:required="true" />

NewbeeFan, 2025-03-24:
> hello! I would like to ask what code you have added and what the detailed setup process is.

Didn't notice the code is not pasted.

I downloaded the sample and open it with Unity 2022.3.58f1. Then I build Camera Viewer scene and adb installed the apk. The player settings of unity are as follow.
The  code I added in AndroidManifest.xml are as follow.
`  <uses-permission android:name="android.permission.CAMERA" />
  <uses-permission android:name="android.permission.ACCESS_MEDIA_LOCATION" />`

[image]

JordiFB, 2025-03-24:
@NewbeeFan  Question about the permissions added in your AndroidManifest.xml.

Android camera permission is required by Unity to use WebCamTexture object, So, this permission is required by our Unity samples.
`<uses-permission android:name="android.permission.CAMERA" /> `

And  in the AndroidManifest.xml , you need to have the specific Meta Horizon camera permission:
`<uses-permission android:name="horizonos.permission.HEADSET_CAMERA" /> `

But, why you added android media location permission? Is it for your specific use case or is required to run our samples in your current configuration?
`<uses-permission android:name="android.permission.ACCESS_MEDIA_LOCATION" />`

swang81, 2025-03-30:
Dear Development Team,

I would like to request some information regarding the use of cameras in **Unreal Engine**. Specifically, could you please provide details on:

1.Camera Usage in Unreal Engine – General guidelines and best practices.

2. Version Dependencies – Any Unreal Engine version requirements or compatibility issues.

3.Implementation Methods – How to properly integrate and use cameras within UE, including any relevant APIs or plugins.

Your insights would be greatly appreciated. Thank you for your support!

closed 2025-04-09 by JordiFB