oculus-samples/Unity-PassthroughCameraApiSamples #5: Passthrough stream as world mesh?

DaveSeidman, 2025-03-25:
I should state right away this isn't an issue or a bug, just a question:

I'm interested in using the live mesh created by the quest using the passthrough cameras. I realize there's a lot of processing happening between the individual camera streams and the final image presented to each eye inside the headset in order to create a world mesh that's stereoscopic. I've been looking for a way to posterize the passthrough feed (reduce the frame rate). 

I could probably leverage this project's samples to persist FLAT images captured from the passthrough cameras but I assume they'd be just that... flat images floating in 3D space at the position and orientation defined by the camera's projection. That won't work for my needs, but if this project exposes access to the 3D textured mesh that would be great. 

Thanks in advance for any response or guidance!

zsp1987, 2025-03-26:
Currently, the Meta only provide APIs that give the global mesh that scanned during the Scene Setup.
And the depth API, which provide the runtime depth Information for ray casting and for render certain effects.

It's possible to combine camera image and depth information to built real-time mesh, but how accurate and performance maybe a concern.

DaveSeidman, 2025-03-26:
I didn't realize the depth information was available, that might be enough to suit my needs when combined with the color camera feed, though I suppose simply mapping the color feed to a plane deformed by the depth information might not line up properly. But thank you for the informaion @zsp1987!

closed 2025-03-31 by zsp1987