oculus-samples/Unity-PassthroughCameraApiSamples #6: Could you provide Camera Usage in Unreal Engine – General guidelines and best practices？

swang81, 2025-03-30:
Dear Development Team,

I would like to request some information regarding the use of cameras in Unreal Engine. Specifically, could you please provide details on:

1.Camera Usage in Unreal Engine – General guidelines and best practices.

Version Dependencies – Any Unreal Engine version requirements or compatibility issues.
3.Implementation Methods – How to properly integrate and use cameras within UE, including any relevant APIs or plugins.

Your insights would be greatly appreciated. Thank you for your support!

JunjieXian, 2025-03-31:
As mentioned above，

I'm really looking forward to it！

swang81, 2025-04-01:
thanks. I am looking for the sample for UE not unity.

JordiFB, 2025-04-11:
Hi @JunjieXian  and @swang81 ,

Unfortunately we don't provide Unreal Engine samples. But, Passthrough Camera Access is going through Android's Camera2 APIs, so you could use Native C++ plugins or the Android Camera module provided by Unreal Engine.

But, I made a test this week using **UE 5 (5.5.3)** + **[Meta XR SDK (74.1)](https://developers.meta.com/horizon/downloads/package/unreal-engine-5-integration/)** and **Android Camera Plugin** provided by UE5:

1. Is possible to access both cameras using **Open URL blueprint module from Media Player**.

2. To construct the URL, you **CAN NOT use the EnumerateWebCamCaptureDevices module**, because it returns 2 front cameras with the same URL (vidcap://front). The UE5 Android Camera implementation stores only 1 id for the front camera (the latest id), so the second camera overwrites the first one.
 
    * To construct the URL, you need to make the URL using a string like this: **vidcap://{CameraIndex}?width={int}?height={int}?fps={int}**
   
    * In my test, Quest Left camera was CameraIndex=0 and the Quest  Right camera was CameraIndex=1.

3. Before access the cameras, **you need to request both permissions** (android.permission.CAMERA and horizonos.permission.HEADSET_CAMERA) using **Request Android permission module** and **user must to accept both**.

    * Also, you need to **add both permission in the AndroidManifest** via UE5 project configuration.

4. Lastly, with **UE5 Android Camera Plugin** you can access the images of both cameras but **NOT get the cameras metadata**. If you need it, you will need to implement your own JAVA calls using C++. (Same as we did for Unity Engine sample in PassthroughCameraUtils c# class.

I hope, this information will help you.

KutlayTunc, 2025-04-16:
Hi Jordy, I was wondering if you are not providing support at all or not in the near future?

JordiFB, 2025-04-21:
Hi @KutlayTunc , I'm focused on Unity Engine support not UE. So, as far as I know about UE support, we are not supporting it in the near future.

EmmanuelEsperanto, 2025-04-21:
> Hi @KutlayTunc , I'm focused on Unity Engine support not UE. So, as far as I know about UE support, we are not supporting it in the near future.

Hi! I tried to do this in Unreal how you said. But I got crush every time when I try access to camera. Can you make simple video how to do that.
Thanks

EmmanuelEsperanto, 2025-04-21:
> Hi [@KutlayTunc](https://github.com/KutlayTunc) , I'm focused on Unity Engine support not UE. So, as far as I know about UE support, we are not supporting it in the near future.

vidcap://1?width=256?height=256?fps=4
This url I got 

[image]
And My engine is 5.5.1 from meta sources of engine and 74.0.0 meta plugin

EmmanuelEsperanto, 2025-04-21:
@JordiFB ping :)

JunjieXian, 2025-04-22:
I try it in UE5.3 and MetaXR SDK(65.0)，but call OpenURL has got false with return.
- I upgrade the HMD OS to  latest and the camera permission has access to the UE-Apk.
- The Android Camera Player Plugin settings have checked all.

Does the MetaXR SDK version must to be latest or there are something I missed?

JordiFB, 2025-04-22:
@JunjieXian  In my previous test using a similar blueprint structure than @EmmanuelEsperanto, it works. Is true that OpenUrl returns false. but I'm able to see the camera image in a widget with a canvas panel + image element.

The Android Camera plugin provided by UE5 is not perfect to work with Meta Quest cameras.

EmmanuelEsperanto, 2025-04-22:
> @JunjieXian  In my previous test using a similar blueprint structure than @EmmanuelEsperanto, it works. Is true that OpenUrl returns false. but I'm able to see the camera image in a widget with a [...]

Oh, do you mean I need to install other Android Camera plugin ? 
I don't understand because I got crash every time when I try OpenURL

JordiFB, 2025-04-22:
@EmmanuelEsperanto No, I used the Android Camera plugin provided by UE5.  

if the OpenURL crashes for you maybe is because you are doing all (check permission and open url) in the Begin Play.

The best way to do it is:

* In Begin Play -> request the permission

* Every tick -> do the following logic:
    * Use a bool variable (IsCameraActive) to open url only 1 time.
    * IF NOT IsCameraActive THEN  {check both permissions} ELSE {do nothing}
    * IF {both permission are granted} THEN 
        * Delay 0.2 seconds 
        * Open the url 
        * Set IsCameraActive to treu ( to not open the url anymore)

EmmanuelEsperanto, 2025-04-22:
> [@JunjieXian](https://github.com/JunjieXian) Seems like your URL is not correctly made, a valid URL will be:
> 
> `vidcap://0?width=1280?height=1080?fps=60`
> 
> Without { }
> 
> I used "append string" to combine the variable values to create the final URL.

[image] I still got crash when I try to OpenURL. I thought it happened because my engine is old or metaxr plugin is old , actually idk. I will try to update metaxr plugin

JordiFB, 2025-04-22:
@EmmanuelEsperanto  The blueprint looks go to me. 

I made my test with UE 5.5.4 and Meta XR 74.1.

And these are the plugins that I have enabled:

* Android Camera Player 
* Android Media Player 
* Android Movie Player
* Android Runtime Permission

Make sure that both permissions are added in the android manifest too (via Project Settings > Android > Extra Permissions)

And review that all required rules are green for Quest 3 in Meta XR Project Setup tool (**Passthrough Camera Access is available only for Quest3 / 3s devices**)

EmmanuelEsperanto, 2025-04-23:
> [@JordiFB](https://github.com/JordiFB) Thx for your help, but it`s not work, and the log shows cant play vidcap://xxx. I`m not sure if the sdk version need the latest. [image]
> 
> And.. Hi [@EmmanuelEsperanto](https://github.com/EmmanuelEsperanto) ,Can you get the camera image to work in UE5.5 + MetaXR (74.1)?

Hi! Now I don't get crash when I try use cameras. Now I can't render the camera image on plane or widgets

JunjieXian, 2025-04-23:
> Hi! Now I don't get crash when I try use cameras. Now I can't render the camera image on plane or widgets

Oh Hi，It's sad, I'm also cant render the camera image on widgets in UE5.3 + MetaXR(65.0), I think it needs latest engine and sdk, but I'm currently unable to upgrade the engine for testing, I'm waiting for your good news 👍

EmmanuelEsperanto, 2025-04-23:
Hello everyone. And know I updated to latest version of meta fork. My branch is oculus-5.5.1-release-1.106.0-v74.1. It's still doesn't works. I just got nothing in my texture

EmmanuelEsperanto, 2025-04-23:
> Hi [@JunjieXian](https://github.com/JunjieXian) and [@swang81](https://github.com/swang81) ,
>
> Unfortunately we don't provide Unreal Engine samples. But, Passthrough Camera Access is going through [...]

Can you share your example project?

DenysKomisar, 2025-04-24:
I have the same problem. The source code from the Metas fork 5.4.4, as well as 5.5.1, does not work with the camera, all permissions are there, but after calling Open URL the application just stops responding and has nothing in the logs

JordiFB, 2025-04-25:
@DenysKomisar  oh! this is unexpected, because camera access was available on v74 as experimental feature. but happy to ear that on v76 is working for you and I hope it works for the other folks.

LarsLeferenz, 2025-04-25:
Sadly for me the update to Horizon OS v76 didn't help. It still crashes on Open URL. I'm running Epics Unreal 5.5.4 and Meta XR Plugin v74.1.  I would also very much appreciate it if anyone could share a basic example project :D .

Otherwise, I might start looking into wrapping the Android Camera2 API sample into an Unreal plugin.

EmmanuelEsperanto, 2025-04-25:
> <oculus-samples/Unity-PassthroughCameraApiSamples/issues/6/2830875158@
> github.com>

Bro, just use EnumerateWebCamera или EnumerateVideoCamera. This topic URL
is wrong for me.
Works only videocap://front
It works for me, for others idk

LarsLeferenz, 2025-04-25:
Both "Enumerate Video Capture Devices" and "Enumerate webcam Capture Devices" also freezes the application for me, so I'm starting to assume there still might be a permissions issue despite "Check Android Permission" showing ´true´ for both HEADSET_CAMERA and CAMERA.

Is there anything I need to configure beyond Unreal? In Horizon OS or Meta Link?

EmmanuelEsperanto, 2025-04-25:
> <oculus-samples/Unity-PassthroughCameraApiSamples/issues/6/2830956982@

Try clear Binaries and Saved, Build, Intermediate
And reboot your headset and try update device software

RaiaN, 2025-05-05:
Just chiming it here, you can use my plugin - [Runtime Video Recorder](https://www.fab.com/listings/36bd136b-2942-4403-9428-96a430ca71ee) from FAB to record in-game videos of your Unreal Engine games. 
It currently supports Render Target recording. Next release (mid-May) will include Camera Feed recording too!

Liyi1998, 2025-06-27:
@JordiFB @JunjieXian @swang81 @LarsLeferenz May I ask if you can help me?

kyrylokuzyk-m, 2025-12-16:
Hey, we released PCA Unreal support in Meta XR Plugin v83: https://github.com/oculus-samples/Unreal-MRUtilityKitSample/commit/9f5cdfda80bc6c1b5d22a8151cf4f0068e66f475