oculus-samples/Unity-PassthroughCameraApiSamples #8: Could not find manifest/application tag in android manifest error

Simo032, 2025-03-31:
When building an Android project, the build fails with the following error from the UpdateManifestWithCodeSample class:

`Could not find manifest/application tag in android manifest.
UnityEngine.Debug:LogError (object)
Meta.XR.Samples.Editor.UpdateManifestWithCodeSample:UpdateManifest (string,System.Collections.Generic.Dictionary`2<string, System.Collections.Generic.HashSet`1<string>>) (at ./Library/PackageCache/com.meta.xr.sdk.core/Scripts/Editor/Util/SampleMetadata/UpdateManifestWithCodeSample.cs:85)
Meta.XR.Samples.Editor.UpdateManifestWithCodeSample:OnPostGenerateGradleAndroidProject (string) (at ./Library/PackageCache/com.meta.xr.sdk.core/Scripts/Editor/Util/SampleMetadata/UpdateManifestWithCodeSample.cs:50)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)`

Despite having a valid AndroidManifest.xml under Assets/Plugins/Android/, the script fails to locate the <application> node during the post-build processing phase.

This is the content of the `Assets/Plugins/Android/AndroidManifest.xml` file:

`<?xml version="1.0" encoding="utf-8" standalone="no"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
          xmlns:tools="http://schemas.android.com/tools"
          android:installLocation="auto">
  
  <application
      android:label="@string/app_name"
      android:icon="@mipmap/app_icon"
      android:allowBackup="false">

    <activity
        android:name="com.unity3d.player.UnityPlayerGameActivity"
        android:theme="@style/Theme.AppCompat.DayNight.NoActionBar"
        android:launchMode="singleTask"
        android:configChanges="locale|fontScale|keyboard|keyboardHidden|mcc|mnc|navigation|orientation|screenLayout|screenSize|smallestScreenSize|touchscreen|uiMode"
        android:excludeFromRecents="true"
        android:exported="true">

      <intent-filter>
        <action android:name="android.intent.action.MAIN" />
        <category android:name="android.intent.category.LAUNCHER" />
        <category android:name="com.oculus.intent.category.VR" />
      </intent-filter>

      <meta-data android:name="unityplayer.UnityActivity" android:value="true" />
      <meta-data android:name="com.oculus.vr.focusaware" android:value="true" />
    </activity>

    <meta-data android:name="com.oculus.ossplash.background" android:value="passthrough-contextual" />
    <meta-data android:name="com.oculus.telemetry.project_guid" android:value="999688c3-c264-4a68-b51b-e2eec0ac83bb" />
    <meta-data android:name="com.oculus.supportedDevices" android:value="quest2|questpro|quest3|quest3s" tools:replace="android:value" />
  </application>

  <uses-feature android:name="android.hardware.vr.headtracking" android:version="1" android:required="true" />
  <uses-permission android:name="com.oculus.permission.USE_ANCHOR_API" />
  <uses-feature android:name="com.oculus.feature.PASSTHROUGH" android:required="false" />
  <uses-permission android:name="com.oculus.permission.USE_SCENE" />
  <uses-permission android:name="horizonos.permission.HEADSET_CAMERA" />
  <uses-permission android:name="android.permission.CAMERA" />
</manifest>
`

Simo032, 2025-03-31:
<?xml version="1.0" encoding="utf-8" standalone="no"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" xmlns:tools="http://schemas.android.com/tools" android:installLocation="auto">
    <application
        android:label="@string/app_name"
        android:icon="@mipmap/app_icon"
        android:allowBackup="false">
        <activity
            android:name="com.unity3d.player.UnityPlayerGameActivity"
            android:theme="@style/Theme.AppCompat.DayNight.NoActionBar"
            android:launchMode="singleTask"
            android:configChanges="locale|fontScale|keyboard|keyboardHidden|mcc|mnc|navigation|orientation|screenLayout|screenSize|smallestScreenSize|touchscreen|uiMode"
            android:excludeFromRecents="true"
            android:exported="true">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />
                <category android:name="android.intent.category.LAUNCHER" />
                <category android:name="com.oculus.intent.category.VR" />
            </intent-filter>
            <meta-data android:name="unityplayer.UnityActivity" android:value="true" />
            <meta-data android:name="com.oculus.vr.focusaware" android:value="true" />
        </activity>
        <meta-data android:name="com.oculus.ossplash.background" android:value="passthrough-contextual" />
        <meta-data android:name="com.oculus.telemetry.project_guid" android:value="999688c3-c264-4a68-b51b-e2eec0ac83bb" />
        <meta-data android:name="com.oculus.supportedDevices" android:value="quest2|questpro|quest3|quest3s" tools:replace="android:value" />
    </application>
    <uses-feature android:name="android.hardware.vr.headtracking" android:version="1" android:required="true" />
    <uses-permission android:name="com.oculus.permission.USE_ANCHOR_API" />
    <uses-feature android:name="com.oculus.feature.PASSTHROUGH" android:required="false" />
    <uses-permission android:name="com.oculus.permission.USE_SCENE" />
    <uses-permission android:name="horizonos.permission.HEADSET_CAMERA" />
	<uses-permission android:name="android.permission.CAMERA" />

</manifest>

Simo032, 2025-03-31:
i'm using the 6000.0.31f1 and i tried to do the clean build of the project

Simo032, 2025-04-02:
If I perform a Build and Run, the build completely crashes and is interrupted. However, if I do Build for Android only, the build is successfully generated. That said, in both cases, the error related to the issue still appears in the console.

JordiFB, 2025-04-03:
@Simo032  based on your last comment, you are able to build for Android but not Build and Run on Android devices (Quest3 or Ques t3s).

Just to double check, Is your project correctly set to Android platform and all the Meta XR checkers are green?

Are you using Oculus XR or Open XR?

Simo032, 2025-04-03:
im using open xr

closed 2025-11-07 by kyrylokuzyk-m