oculus-samples/Unity-PassthroughCameraApiSamples #10: accessing both cameras

jkim447, 2025-04-05:
Hi, thank you very much for the open source.

I'm curious -- is there a workaround for accessing both cameras at once? We are trying to explore stereo matching problem using Quest device and having access to both camera images in real-time is critical. I know this is mentioned as a limitation of the feature but would you mind letting me know if there's a hack or some unofficial workaround strategy to acess both cameras at once? Thank you in advance.

JordiFB, 2025-04-07:
Hi @jkim447  our current integration with Unity Engine is through **WebCamTexture object** provided by Unity.  Unfortunately, [WebCamTexture Object only provides access to 1 camera on Android and IOS platforms](https://docs.unity3d.com/6000.0/Documentation/ScriptReference/WebCamTexture.html). It means that is NOT possible to create 2 WebCamTexture objects pointing to 2 different Cameras.

jkim447, 2025-04-07:
are there plans to enable access to both cameras at the same time? Thanks in advance.

JordiFB, 2025-04-08:
Our team is look into it, but we don't have any estimation time about when it will be available. 

Note: You always have the option of writing your own Android Camera2 JNI wrapper for Unity, It will give you access to both camera images at the same time.

jkim447, 2025-04-08:
Thank you for the feedback. Would you happen to have some sample code that we can try to extend so that we can get both camera access?

JordiFB, 2025-04-11:
@jkim447  You could check the [Native Samples](https://github.com/oculus-samples/Unity-PassthroughCameraApiSamples/tree/main/NativeSamples) code to see how access the camera using Android native code.

jkim447, 2025-04-13:
thank you. Any chance the team could develop a library where we could call from Unity for both RGB camera access simultaneously?

I think this could be super helpful to the community, I would feel that the use-case for having access to RGB camera will be quite broad.

JordiFB, 2025-04-21:
@jkim447 Thanks for your feedback. We will add your request in our list of improvements for this feature.

wetoo-cando, 2025-05-25:
> [@jkim447](https://github.com/jkim447) You could check the [Native Samples](https://github.com/oculus-samples/Unity-PassthroughCameraApiSamples/tree/main/NativeSamples) code to see how access the camera using Android native code.

The link does not work …

kyrylokuzyk-m, 2025-11-07:
Hey, we recently introduced a new PassthroughCameraAccess component in MRUK v81. To access both camera at the same time, create two PassthroughCameraAccess components with different CameraPosition: the first one with CameraPositionType.Left and the second one with CameraPositionType.Right.

Please also see this [migration guide](https://developers.meta.com/horizon/documentation/unity/unity-pca-migration-from-webcamtexture) from the old WebCamTextureManager and an [API reference](https://developers.meta.com/horizon/reference/mruk/v81/class_meta_x_r_passthrough_camera_access).

closed 2025-11-07 by kyrylokuzyk-m