oculus-samples/Unity-PassthroughCameraApiSamples #12: Request: Unity 6 tag

bradley-newman, 2025-04-15:
The main branch is configured for Unity 2022.3.52f1. Upgrading to Unity 6000.0.38f1 is a non-trivial process involving many steps and modifications to many files. Even after closely following the docs, it's not clear whether my project config is valid. It would be really helpful if Meta provided a tag for Unity 6000.0.38f1 where the project is pre-configured.

zsp1987, 2025-04-16:
Thanks for the feedback, We test the project should run with Unity 6. 
It will be hard to tag with each unity version, but we may consider tag with certain Unity 6 version in the future.

kyrylokuzyk-m, 2025-11-07:
We updated the project to Unity 6 recently. Thank you for flagging the issue!

closed 2025-11-07 by kyrylokuzyk-m