oculus-samples/Unity-PassthroughCameraApiSamples #13: Image quality...

smallfly, 2025-04-16:
Hello,

The image returned by Unity's WebCamTexture appears to be of lower quality compared to the actual passthrough, even when capturing at 1920x960 resolution. Is this a known limitation of WebCamTexture, and could using the native Camera2 API provide better image quality?

Thanks

smallfly, 2025-05-02:
Sorry for the late reply, and thanks for your response. 
But, how is the image quality you’re getting with WebCamTexture?

zsp1987, 2025-05-21:
For webcamtexture the max resolution is 1280x960

zsp1987, 2025-06-02:
Hi, it's a known issue, and we are going to improve the image quality with antialias in our frag shader.

johanneskopf, 2025-07-05:
@zsp1987 The image is particularly distorted on the right side. Interestingly, it does not matter whether the image is taken with the right or the left eye. Do you have any idea what causes this effect, which mainly occurs on the right side? The left side of the image is much sharper/anti-aliased.

johanneskopf, 2025-07-29:
My Quest 3 was updated to the latest firmware (v78.1030) today, and the image distortion has now disappeared. So it seems that this issue has been fixed with the latest firmware. @smallfly , can you confirm this?

Next, it would be great to have access to a higher resolution/frame rate, as the image quality is still far from the passthrough image.

smallfly, 2025-08-22:
Just saw that news on LinkedIn. 
I will test and report back. 
Right now my Quest 3 is already updating to the v79.

closed 2025-11-07 by kyrylokuzyk-m