oculus-samples/Unity-PassthroughCameraApiSamples #16: built six scenes at the same time?

XuanbingXie, 2025-05-02:
I really appreciate your work, but I encountered a problem. When I built six scenes at the same time, I pressed the permission twice, but it still showed that I did not allow it. What's going on? Or should I build the six scenes separately? Thank you

JordiFB, 2025-05-02:
Hi @XuanbingXie , What do you mean with "built six scenes"? Could you give us more details (unity version, horizon os version, etc)? Are you seeing any message in the console log?

XuanbingXie, 2025-05-02:
I also tried to build only the cameratoworld project, and found that after pressing the consent button twice, it still showed permission not granted.My unity version is 6000.0.48f1, use meta quest 3, and there is no warning and console log.

XuanbingXie, 2025-05-02:
I encountered the same problem as #3, after permission, it still appear permission not granted, so bad

JordiFB, 2025-05-02:
@XuanbingXie  Very strange behavior. If you build and run the samples are you experiencing the same issue?

Could you capture some logs using "adb logcat >> log.txt" and share it with us?

Thank you

JordiFB, 2025-06-06:
@XuanbingXie Any update on your issue? If you found the solution to it, we can close this issue. Thank you.

XuanbingXie, 2025-06-07:
> <oculus-samples/Unity-PassthroughCameraApiSamples/issues/16/2949134515@
> github.com>

Thank you for your email!Actually the problem is still not solved, but we
changed our thinking and did not use your open source. Thank you.

closed 2025-06-12 by JordiFB