oculus-samples/Unity-PassthroughCameraApiSamples #22: About Parameters From Camera and Usage of the PassthroughCameraUtils

OrangesFlower, 2025-05-18:
Hi developers,

I'm implementing a function about point at image of camera and then locate it in scene, but I meet some problems.

Refering to this module: [PassthroughCameraUtils.cs](https://github.com/oculus-samples/Unity-PassthroughCameraApiSamples/blob/9c9ffc4409a35b2d262c062c871cd676fbbc7817/Assets/PassthroughCameraApiSamples/PassthroughCamera/Scripts/PassthroughCameraUtils.cs), I want to build my own point to world system, and I get camera parameters, mannually calculate the Ray. So I implemented this:

```cs
using UnityEngine;
using System.Collections.Generic;
using PassthroughCameraSamples;

public class CameraParameterSnapshot {
  public Vector2Int Resolution { get; private set; }
  public Vector2 FocalLength { get; private set; }
  public Vector2 PrincipalPoint { get; private set; }
  public float Skew { get; private set; }
  public Vector3 Position { get; private set; }
  public Quaternion Rotation { get; private set; }

  public CameraParameterSnapshot(PassthroughCameraEye cameraEye) {
    // get resolution
    List & lt;
    Vector2Int & gt;
    outputSizes = PassthroughCameraUtils.GetOutputSizes(cameraEye);
    Resolution = outputSizes.Count & gt;
    0 ? outputSizes[0] : new Vector2Int(0, 0);

    // get intrinsics
    var intrinsics = PassthroughCameraUtils.GetCameraIntrinsics(cameraEye);
    FocalLength = intrinsics.FocalLength;
    PrincipalPoint = intrinsics.PrincipalPoint;
    Skew = intrinsics.Skew;

    // get pose
    var pose = PassthroughCameraUtils.GetCameraPoseInWorld(cameraEye);
    Position = pose.position;
    Rotation = pose.rotation;
  }

  public Ray ScreenPointToWorldRay(Vector2Int screenPoint) {
    // 1. calculate camera space direction
    var directionInCamera =
        new Vector3((screenPoint.x - PrincipalPoint.x) / FocalLength.x,
                    (screenPoint.y - PrincipalPoint.y) / FocalLength.y, 1f);

    // 2. transform to world space
    var directionInWorld = Rotation * directionInCamera.normalized;
    return new Ray(Position, directionInWorld);
  }
}
```
Usually, by this way I only need to get the point on the images and call `ScreenPointToWorldRay` function, and then I can get the Ray in the world space. But something go wrong.

[image]

(px, and py are the loaction of point in canvas)

As you see, when I use RayInteractor point on the **center** of the frame, but the **gray sphere** did not appeared in correct place of real world (It should be around the point place in the real world by the Depth raycasting function). I print the resolution and intrinsics of the camera on debug text, and I find the camera resolution is (320×240), same as my canvas so it is correct. but when I check the principal point it was a wierd number (642.11, 480.76), it seems doubled resolution. In my mind, principal point always in the frame and the ray should be calculate like this:

[image]

Obviously it exceed the frame size, so is this a problem? And is the method I used correct?

Thanks for your help, any help will be appreciate.

OrangesFlower, 2025-05-19:
I find the point, it is not about the principal point ---- It's about the camera resolution. I think it should be 1280×960, but now is 320×240, but why?

JordiFB, 2025-05-20:
Hi @OrangesFlower, just to double check your configuration. 

Do you set the request resolution to 1280x960 in the configuration prefab (**WebCamTextureManagerPrefab**) ?

Also, could you share your Unity project details ( Unity version, current OS, ...) ? 

Thank you,

JordiFB, 2025-05-20:
@OrangesFlower  Thanks for the information. 

Seems like the error could be in how you get the **resolution**.

In your code, you get the list of all available resolution using  **PassthroughCameraUtils.GetOutputSizes(cameraEye);**. But you select the first value from the list **outputSizes[0]**.

I guess, the first value is the lowest resolution, not the highest.

Check how we select the resolution in **WebCamTextureManager: InitializeWebCamTexture()** function.

* We order the resolutions and get the last value.

closed 2025-06-06 by OrangesFlower