oculus-samples/Unity-PassthroughCameraApiSamples #25: Full passthrough capture for blur effect

hvp, 2025-05-21:
Hey there, 

Thanks so much for the sample code! This might be a silly question, as I'm aware there's a max resolution limitation, but I'm trying to achieve a full screen effect where I blur the entire field of view.

I'm using the CameraTextureMapper class to push my camera intrinsics to my shader and it seems that the when the computed UVs come back at the extremes, I get stretching effects. I.e. there's no resolution / coverage of that part of the composited passthrough image?  

Here's the snippet here (based off the blur shader that xrdevrob built, which I think came from initial frosted glass shader?) 

` float3 diff = IN.worldPos - _CameraPos;
 float3 localPos = mul(_CameraRotationMatrix, float4(diff, 1.0)).xyz;
 if (localPos.z < 0.001)
     discard;

 float uImage = _FocalLength.x * (localPos.x / localPos.z) + _PrincipalPoint.x;
 float vImage = _FocalLength.y * (localPos.y / localPos.z) + _PrincipalPoint.y;

           
 float scaleX = _MainTex_TexelSize.z / _IntrinsicResolution.x;
 float scaleY = _MainTex_TexelSize.w / _IntrinsicResolution.y;
 uImage *= scaleX;
 vImage *= scaleY;

 float2 computedUV = float2(uImage / _MainTex_TexelSize.z, vImage / _MainTex_TexelSize.w);

 computedUV = ClampUV(computedUV);

   if (computedUV.x < 0.05 || computedUV.x > 0.95 || computedUV.y < 0.05 || computedUV.y > 0.95)
 {
     discard;
 }`

I did try and cycle the cameras left and right and recomposite the combined image into one larger texture, but I'm afraid I've hit the limit to my understanding of how to re-composite the webcam textures! And I'm assuming we're not even getting the full resolution out of the cameras any way so it might not be possible at all? 

Any insight would be greatly appreciated even it's just to confirm that it's not currently possible!

Thanks a bunch, 
Harrison

JordiFB, 2025-05-22:
@hvp your project is very cool but unfortunately Passthrough Camera Access API has not been designed to replace the Passthrough composite image.  Is not currently possible because of the image max resolution and the fact that the camera images are pre-processed internally  to remove the eye distortion.

hvp, 2025-05-22:
Thanks so much for confirming!