oculus-samples/Unity-PassthroughCameraApiSamples #31: Maximum frame rate

johanneskopf, 2025-07-10:
Hi,

According to the [documentation](https://developers.meta.com/horizon/documentation/spatial-sdk/spatial-sdk-pca-overview/) the data rate for the RGB images supplied by the PCA is 30Hz. Is this limited by the hardware or a limitation of the current software implementation and can therefore be improved in the future? It would be great if 60Hz was available.

Thanks for your work on this API so far!

zsp1987, 2025-07-11:
It's software limitation.

We will check if this can be improved in the future.

johanneskopf, 2025-07-14:
Thank you for the fast reply.

Awesome, glad to hear that!

dudziakl, 2025-12-17:
> Hey, 60Hz is now available since MRUK v83.

OMG, I should start to believe in Santa again 🎅