oculus-samples/Unity-PassthroughCameraApiSamples #32: How to apply another object detection Yolo model.

Best-Gagil, 2025-08-14:
Dear Repository Team

I hope this message finds you well. My name is Guillermo, and I’m currently working on my thesis project using the Meta Quest 3S and Unity. A key component of my project involves detecting a custom-built exoskeleton composed of three main parts: Hand, Micro, and Button.

I discovered your repository while exploring solutions for multi-object detection in Unity, and I must say the MultiObjectDetection scene works impressively well! Inspired by its performance, I decided to train my own YOLO-based model specifically tailored to detect the components of the exoskeleton.

So far, I’ve trained models using YOLOv5, YOLOv9, and YOLOv11, all implemented in PyTorch. The results have been promising. I then exported the YOLOv11 model to ONNX format, which also performs well. The next step was converting the ONNX model to a Sentis-compatible format for use in Unity.

Here’s where I encountered some issues:

I attempted to use the export (ONNX to Sentis) functionality provided in the MultiObjectDetection scene, but unfortunately, nothing happened when I selected my model—no errors or warnings were displayed.
As an alternative, I used the Onnx2Sentis application from another repository to perform the conversion. This process completed successfully.
I updated the Sentis object in Unity with my new model and replaced the classes.txt file with one matching my order used during training.
However, when I build and deploy the project to my Meta Quest 3S, the application runs smoothly, but the detection does not work. Interestingly, when I revert to the original model and class file provided in the repository, everything functions perfectly again.

To help you better understand my setup, here are some details:

* Unity version: 2022.3.58F1
* Sentis version: 2.1.1
* Device: Meta Quest 3S

Here I have an image example of my ONNX model detection:

[image]

I would greatly appreciate any advice or insights you could share regarding the integration of custom YOLO models into the MultiObjectDetection scene. Specifically, I’m curious if there are additional steps or configurations I might be missing when replacing the model and class file.

Thank you very much for your time and for the incredible work on this project. I look forward to hearing from you!

Warm regards,
Guillermo Aguilera

jojorabbit, 2025-09-06:
I have similar problem, from what i investigated in logcat seems the model in this sample uses 2 outputs, and model that i trained has 1 output, so maybe yours also has one output. Check script SentisInferenceRunManager it asks for 2 outputs one with m_engine.PeekOutput(0) and another in m_engine.PeekOutput(1) to get the label.
Hope it makes sense.

Best-Gagil, 2025-12-07:
Yes! It was solved! and It was the same topics that you @metaRobi and @jojorabbit mentioned ! Thank you very much for your answers.
PD. Sorry for the late answer.